QUT-Motorsport/QUTMS_Driverless #12: Figure out a good way to mount in /dev/sbg so that it doesnt create an empty folder when the device isnt present

alistair-english, 2021-11-21:


b1n-ch1kn, 2022-01-03:
does this need to be looked at again? or will robostack fix?

alistair-english, 2022-01-03:
Nah, RoboStack won't fix, this is a docker compose thing, probably won't bother looking at it until we want to use the SBG again

alistair-english, 2022-03-17:
This is relevant again now as we are planning on using the SBG usb interface + ROS node when running driverless. Good issue for if we want someone to learn more about the docker infrastructure.

b1n-ch1kn, 2022-07-03:
We want to do IMU+lidar data collection in the carpark early this week while liam is in. Any new ideas for this?

alistair-english, 2022-07-03:
> We want to do IMU+lidar data collection in the carpark early this week while liam is in. Any new ideas for this?

We can sort this out tomorrow (Monday) - I think there's a straightforward way to do this, just need to do some googling.

closed 2022-07-06 by b1n-ch1kn