QUT-Motorsport/QUTMS_Driverless #25: Update ROS installation in docker to RoboStack

alistair-english, 2021-12-02:
https://github.com/RoboStack

Should mean we can unify our installation of ROS again cause RoboStack supports ARM

b1n-ch1kn, 2021-12-09:
Some galactic packages for ARM aren't in Robostack yet in like vision (for cv_bridge), and velodyne drivers aren't even available for linux :(

[image]

[image]

b1n-ch1kn, 2021-12-09:
given this, should we not use dockerhub for images?

alistair-english, 2021-12-09:
> Some galactic packages for ARM aren't in Robostack yet in like vision (for cv_bridge), and velodyne drivers aren't even available for linux :(
> 
> [image]
> 
> [image]

https://robostack.github.io/galactic.html has CV deps on ARM (which should be all we need - hence why we need to upgrade to galactic first). Plus, we can always ask for packages we want to be added - pretty sure they just havent had much ARM usage so havent bothered  adding packages.

alistair-english, 2021-12-09:
> given this, should we not use dockerhub for images?

wdm by this?

b1n-ch1kn, 2021-12-09:
Ok, sounds good.

Yesterday you mentioned how we'd have a system where the images will be built, uploaded to docker hub, and pulled on the car, so I made a task: https://github.com/QUT-Motorsport/QUTMS_Driverless/issues/34

alistair-english, 2021-12-09:
https://vimeo.com/649654404

b1n-ch1kn, 2021-12-14:
I like the yaml file for all the dependancies.
Link for dependancies we require:
https://robostack.github.io/galactic.html

Also, do we have to add to the yaml file everything thats in each package.xml file? or can we install rosdep to source it automatically?

b1n-ch1kn, 2021-12-14:
Will turn the `conda_instructions.txt` into wiki page too

closed 2022-01-03 by b1n-ch1kn