QUT-Motorsport/QUTMS_Driverless #33: FIgure out "deployment" architecture for Jetson

alistair-english, 2021-12-07:
https://github.com/ultralytics/yolov5/issues/5705#issuecomment-972901057 -> "These devices are really only meant as deployment destinations for already trained models. Your workflow would be to train a model in the cloud or with a local desktop GPU, then export to a deployment format like TRT and deploy to Jetson."

Attempted to get the current stack running on the Jetson but ran into many versioning issues caused by its ARM processor (a bunch of python dependencies needed to be compiled from scratch, etc).

Want to figure out a cleaner way to deploy models to the car.

b1n-ch1kn, 2021-12-07:
Nvidia says they have wheels build for arm
https://forums.developer.nvidia.com/t/pytorch-for-jetson-version-1-10-now-available/72048

b1n-ch1kn, 2021-12-07:
Ah ha
The URL where torch 1.10.0 is currently coming from is wrong
[image]

Should be:
https://nvidia.box.com/shared/static/fjtbno0vpo676a25cgvuqc1wty0fkkg6.whl

Also, do we need cython?
[image]

alistair-english, 2021-12-07:
Yeah the pytorch wheel works, the issue is with the python dependencies of yolov5 - it require versions of packages higher than what's available on apt so pip has to build them. Yesterday I think it was failing on building matplotlib, and id sunk a day and a half into it without success so I had to move on.

I think if we spent another day or so on it, we could probably make it work but given that takes us past the demo, I think we're better off figuring out a "cleaner" solution that involves running an exported model of some sort, which will hopefully require minimal dependencies.

b1n-ch1kn, 2021-12-07:
Ah righto, the vision retirements.txt is a cutdown version of what the yolo model required when I went through and worked out what wasn't needed. Matplotlib isn't:
[image]

b1n-ch1kn, 2021-12-07:
Actually, that might not be true. something else might have installed it automatically when i was testing

alistair-english, 2021-12-08:
https://blog.roboflow.com/deploy-yolov5-to-jetson-nx/

This article says to just install all the Yolo repo requirements - maybe something along those lines is the way to go...

b1n-ch1kn, 2021-12-08:
they have a pip package too:
pip install yolov5 will do all the requirements

b1n-ch1kn, 2021-12-08:
> https://blog.roboflow.com/deploy-yolov5-to-jetson-nx/
> 
> This article says to just install all the Yolo repo requirements - maybe something along those lines is the way to go...

This uses the pytorch container as base so maybe that has extra stuff needed for the pip wheels

b1n-ch1kn, 2021-12-09:
https://catalog.ngc.nvidia.com/orgs/nvidia/containers/l4t-pytorch
Pytorch nvidia container that we could build on?

alistair-english, 2021-12-09:
Yep, I'm looking at upgrading our dockerfiles to use robostack (https://github.com/QUT-Motorsport/QUTMS_Driverless/issues/25). This should mean we can unify our dockerfiles again (no more special jetson containers - robostack supports ROS on ARM) so as part of that I can look at making the jetson version of the build come off that pytorch container.

b1n-ch1kn, 2021-12-09:
If https://github.com/QUT-Motorsport/QUTMS_Driverless/issues/25 is aimed to fix yolov5 model stuff, is there a way to merge this issue into it?

b1n-ch1kn, 2022-01-03:
Wasn't quite solved with conda packages (arm be like)
Used a prebuild container for ros to be built from source on a pytorch container

closed 2022-01-03 by b1n-ch1kn