QUT-Motorsport/QUTMS_Driverless #34: Docker hub images

b1n-ch1kn, 2021-12-09:
Get good with using Docker so we can build images then deploy them to Dockerhub so the car can just pull them in without building containers on the car (ideally). Saves time waiting with the car / needed to power it up.
Would have all dependencies for software running on the Jetson and ROSCube so we can just run the container.

alistair-english, 2021-12-09:
This might be a bit of a longer term goal - at the moment we still haven't settled on a complete architecture, so adding extra build steps may prove more cumbersome than helpful.

However, this is still relevant as a future goal, since we will always have custom build steps in our containers.

b1n-ch1kn, 2022-07-07:
this probably doesnt need to be done now that we're just pulling existing base docker images anyway and adding conda

alistair-english, 2022-07-08:
yeah this is no longer something we need to pursue

closed 2022-07-09 by b1n-ch1kn