QUT-Motorsport/QUTMS_Driverless #37: Building Images ip instructions should be more specific

TheEppicJR, 2021-12-12:
> and copy the value next to 'IPV4 Address'.

should be 

> and copy the value next to 'IPV4 Address' under 'Ethernet adapter vEthernet (WSL)'.

b1n-ch1kn, 2021-12-13:
Should be this address:

[image]

Though, I haven't tested whether the "Ethernet adapter vEthernet (WSL)" IPv4 address works too

alistair-english, 2021-12-13:
I believe both will work

b1n-ch1kn, 2021-12-13:
In which case, did we want to change the .env lines that require user input from `{IPV4 HERE} ` to `export WSL_HOST=$(ip route | awk '{print $3; exit}')` ? i believe this automatically gets the vEthernet WSL

TheEppicJR, 2021-12-13:
I am on wifi and have "Wireless LAN adapter Wi-Fi" which does not work for me, but that may just also be that my LAN is a bit sus, either way I believe that pointing it to the vEthernet WSL is a better default.

alistair-english, 2021-12-14:
> In which case, did we want to change the .env lines that require user input from `{IPV4 HERE}` to `export WSL_HOST=$(ip route | awk '{print $3; exit}')` ? i believe this automatically gets the vEthernet WSL. btw this was just copied from my ubuntu bashrc file but we'd change it to `SIM_HOST=` not `export WSL_HOST=`

I attempted to add a subshell (the `$(...)`) into the envfile before but makefiles dont support it.

I see you've updated the wiki already @b1n-ch1kn so will close this issue.

closed 2021-12-14 by alistair-english

b1n-ch1kn, 2021-12-14:
ah oh well, for now a fixed IP works