QUT-Motorsport/QUTMS_Driverless #43: Get sbg_driver and velodyne packages onto robostack

alistair-english, 2021-12-17:
We are moving towards using robostack to manage our ROS environment, at the moment, ROS 2 packages we use `sbg_driver` and `velodyne` are not on robostack.

https://robostack.github.io/Contributing.html

b1n-ch1kn, 2021-12-18:
So i got the sbg driver to make me a recipe then build
[image]

just not sure where the built stuff went lol
[image]

b1n-ch1kn, 2021-12-20:
SBG driver submitted as a pull request to [robostack ](https://github.com/RoboStack/ros-galactic/pull/65) (lets see how it goes)

b1n-ch1kn, 2021-12-20:
Not having any more luck with velodyne_pointcloud - all other velodyne packages work (msgs, laserscan, driver) but to get our the velodyne driver outputting we need the pointcloud package.

Gets to here:

[image]
[image]
[image]

[CMakeError.log](https://github.com/QUT-Motorsport/QUTMS_Driverless/files/7743522/CMakeError.log)
[CMakeOutput.log](https://github.com/QUT-Motorsport/QUTMS_Driverless/files/7743524/CMakeOutput.log)

Tobias-Fischer, 2021-12-21:
Hiya, for you guys I'll be happy to have a look once I'm back from leave on 4 Jan. Please ping me then.

Tobias-Fischer, 2021-12-21:
That's actually an easy one, add two lines like that for the point cloud package: https://github.com/RoboStack/ros-galactic/blob/06f1f91b1c35896847bc48839103fa6b092d4079/patch/dependencies.yaml#L6-L7

b1n-ch1kn, 2021-12-22:
> That's actually an easy one, add two lines like that for the point cloud package: https://github.com/RoboStack/ros-galactic/blob/06f1f91b1c35896847bc48839103fa6b092d4079/patch/dependencies.yaml#L6-L7

Yep this worked. Thanks again!

b1n-ch1kn, 2021-12-22:
Thanks for your help today @Tobias-Fischer 
Looking forward to seeing our system come together on RoboStack soon.

closed 2021-12-22 by b1n-ch1kn