QUT-Motorsport/QUTMS_Driverless #44: lidar_pipeline2: lidar_perception.py

liamferrante, 2021-12-20:
Investigate a suitable size for the Message Subscription queue.
Until the pipeline is running faster than the LiDAR camera, the queue size will only ever increase in size. 

Currently, the queue size is set to 2.
In the LiDAR pipeline 1.0 this value is 10.

b1n-ch1kn, 2021-12-20:
One thing to watch is the quality of service (QoS) of the published node. Often, the subscriber has to match the publisher to ensure they talk nicely. TBH not sure how queue size will affect QoS.

Anyway heres a link for some more reading.
https://stackoverflow.com/questions/56444248/reason-to-set-queue-size-of-ros-publisher-or-subscriver-to-a-large-value

b1n-ch1kn, 2022-08-01:
so reading into this for whatever reason
we dont want a big queue, we want the most up to date detection - so once its finished processing a scan, it'll immediately get the last published msg. maybe a queue of 2 max? so it wont be waiting around for a msg if somehow its faster and out of sync with velodyne.
big queue of 20 would mean it has to get thru the 20 stored before it gets to a new, more recent pointcloud scan

closed 2022-08-05 by b1n-ch1kn