QUT-Motorsport/QUTMS_Driverless #45: Migrate ZED ROS2 wrapper to Galactic

b1n-ch1kn, 2021-12-22:
Currently, the ZED wrapper is made for Foxy (no package anyway, just built from source).
Need to move to Galactic if we want the package made for RoboStack environment.

For now we use the wrapper in a docker container (existing solution) that publishes topics for the processing node in the RoboStack env to subscribe to.

b1n-ch1kn, 2022-01-03:
Also saw this dockerfile uses the same ros from source as the one built on pytorch. Instead of foxy, we could just use the galactic one (provided the zed wrapper doesnt break)

alistair-english, 2022-01-03:
Unfortunately zed wrapper doesn't support galactic out of the box - had a stab at it with the fork in the QUTMS organisation, not sure how far off I was

closed 2022-08-29 by b1n-ch1kn