QUT-Motorsport/QUTMS_Driverless #50: Get TRT running on L4T

b1n-ch1kn, 2022-01-14:
With the success of TensorRT optimisation, and significantly less overhead than the PyTorch requirements, we can try getting TRT using a conda env from the dockerfile.base and install Pycuda + scipy (the *only* TRT requirements)

b1n-ch1kn, 2022-01-15:
Forgot that this L4T runs python3.6 not python3.8 which conda runs in rip
might need more digging eventually

b1n-ch1kn, 2022-01-15:
https://forums.developer.nvidia.com/t/how-to-support-python3-8/168813
https://arcanesciencelab.wordpress.com/2020/05/24/building-python-3-8-3-on-jetpack-4-4-dp-and-the-jetson-xavier-nx/

alistair-english, 2022-01-19:
This done?

b1n-ch1kn, 2022-01-19:
No its cooked, proprietary nvidia container stuff for python tensorrt. can just pip install tensorrt alongside pycuda. might have to run in it's container not a conda env