QUT-Motorsport/QUTMS_Driverless #58: SBG CAN Messages

alistair-english, 2022-01-19:


b1n-ch1kn, 2022-01-19:
Possibly blocked by https://github.com/QUT-Motorsport/QUTMS_Driverless/issues/12 ?

TomFraser, 2022-01-19:
> Possibly blocked by #12 ?

Nope, this just refers to adding the Composers & Parsers for the SBGs ECAN protocol, #12 refers to the ECOM protocol which is over Serial.

TomFraser, 2022-01-25:
Investigating this, which specific information would you guys like from the SBG (position, velocity, acceleration, UTC time etc)? Do we have topics planned for passing the information onto? @alistair-english @b1n-ch1kn

alistair-english, 2022-01-25:
https://github.com/SBG-Systems/sbg_ros2_driver

Have a look and see what message types this one uses. Haven't got any plans for what exact data we want but if we can keep it flexible that would be good.

closed 2022-03-25 by TomFraser