QUT-Motorsport/QUTMS_Driverless #65: Compute Enclosures and Mounting

alistair-english, 2022-01-19:
For George

b1n-ch1kn, 2022-01-19:
+ waterproof

alistair-english, 2022-01-25:
Compute Box Requirements
- Cooling - fans (can do scoop too, but need to have cooling when car is static for long periods)
- Mounted to roll hoop
- Waterproof
- Clear top? / Latching top
- Able to add buttons to the side (probably still good if we can do this even with latching top)
- Grommets for wires
- UV resistance, will be out in the sun a lot

Fits in it:
- ROScube-I (core only - no expansion, https://www.adlinktech.com/Products/Download.ashx?type=MDownload&isDatasheet=yes&file=1785%5cROScube-I-20201116.pdf)
- Jetson AGX Xavier Developer Kit (https://developer.nvidia.com/embedded/jetson-agx-xavier-developer-kit)
- EBS Controller (PCB - assume 120mm x 120mm x 40mm)
- CAN Tranciever (USR-CANET200 https://www.iot-store.com.au/products/industrial-can-bus-to-ethernet-converter-w-rs485-port-usr-canet200)
- Anything else we've forgotten, try and get the biggest box that fits within the rollhoop area so we have flexibility

Connections:
- 2x Ethernet
- Heaps of wires (power, sensors)
- Just need clearance on the edges of the box for a lot of ad-hoc grommets

GeorgyUshakov, 2022-02-01:
The Two-Box Solution.

Demonstrating that all the non-ROScube components can fit inside a 30x30x10 box:
[image]

[TheBox 2.zip](https://github.com/QUT-Motorsport/QUTMS_Driverless/files/7977256/TheBox.2.zip)

GeorgyUshakov, 2022-02-08:
### Some options for TheBox™es:
**Plastic boxes - most are IP66 or thereabouts  which is like the highest grade of protection (or even above that which means we're good on the waterproofing front; 1st digit represents solid protection and 2nd digit represents liquid protection):** 
Brenclosures: https://www.brenclosures.com.au/plastic-enclosures/
Polynova PC: 270x270x170, IP66
Polynova PM: 270x270x165, IP65
Polynova PN: 360x250x150, IP67

RS PRO Grey Polycarbonate General Purpose Enclosure, IP65, 344 x 289 x 117mm: $121.04; https://au.rs-online.com/web/p/general-purpose-enclosures/1972019

RS PRO Grey Fibreglass Enclosure, IP67, 300 x 300 x 175mm: $231.85; https://au.rs-online.com/web/p/general-purpose-enclosures/0507999

RS PRO Grey PET Enclosure, IP66, 255 x 250 x 120mm: $228.86; https://au.rs-online.com/web/p/general-purpose-enclosures/9188825

GRP IP68 Grey Terminal Enclosure 250H x 255W x 125D:  $71.34; https://www.tro.com.au/GRP-IP68-Grey-Terminal-Enclosure-250H-x-255W-x-125D

**Metal boxes - more expensive and with less options found so far:**
Aluminium Electrical Enclosure 300H x 300W x 150D IP66 Grey: $269.85; https://www.tro.com.au/Aluminium-Electrical-Enclosure-300H-x-300W-x-150D-IP66

Aluminium Electrical Enclosure 300H x 250W x 150D IP66 Grey:  $260.70; https://www.tro.com.au/Aluminium-Electrical-Enclosure-300H-x-250W-x-150D-IP66

RS PRO Silver Die Cast Aluminium Enclosure, IP66, Shielded, 250 x 250 x 100mm: $90.05; https://au.rs-online.com/web/p/general-purpose-enclosures/5173428
^ wouldn't fit ROScube but very cheap

Note: a 25x25 enclosure would need to stack some components on top of each other in order to fit a fan (eg. CAN to Ethernet on top of Jetson as below), although a deeper box may use the extra depth for the fan instead.
[image]

Note note: I presume we can just cut the fan exhaust hole/s into the box ourselves - would be best in terms of flexibility in box choice and component arrangement, but may lead to waterproofing issues (I saw shrouds mounted where the exhaust is and the thing was called weatherproof so that might be enough).

b1n-ch1kn, 2022-02-08:
Neat, these look like good options with sufficient mounting options/

Metal boxes might get toasty for the computers in the sun lol.

b1n-ch1kn, 2022-02-10:
Initial harness diagram for routing cables in and out of boxes

[image]

b1n-ch1kn, 2022-02-10:
Was looking into the:
Polynova PM 270x270x165 https://brenclosures-webxopt.netdna-ssl.com/wp-content/uploads/Product-Leaflet-Technical-Polynova-PM.pdf
And:
Polynova PC 270x270x170+42 https://brenclosures-webxopt.netdna-ssl.com/wp-content/uploads/Product-Leaflet-Technical_Polynova-PC-2021.pdf

They look like good options, but cant see prices for them as @GeorgyUshakov noted or at other retailers.

So I found some more options at good prices:
Fibox 300x230x110 - $132.06: https://au.rs-online.com/web/p/general-purpose-enclosures/2189896
Fibox 278x278x130 - $131.34: https://au.rs-online.com/web/p/general-purpose-enclosures/1943134
Fibox 278x278x180 - $119.08 (lol 50mm taller but $12 cheaper): https://au.rs-online.com/web/p/general-purpose-enclosures/1943077

b1n-ch1kn, 2022-02-10:
TLA test sizes

- Fibox 278x278x180 - $119.08:
[image]
[image]

- Fibox 300x230x110 - $132.06:
[image]
[image]

- Polynova PC 270x270x170 - $165:
[image]
[image]

- Tro 300x230x130 - $42.25:
[image]
[image]

b1n-ch1kn, 2022-02-11:
Contacted Tro Pacific to get datasheets for:
https://www.tro.com.au/Polycarbonate-Enclosure-Clear-Lid-IP66-300H-x-230W-x-130D 

[Enlec 300x230x130.pdf](https://github.com/QUT-Motorsport/QUTMS_Driverless/files/8046747/Enlec.300x230x130.pdf)
[image]

<br>

https://www.tro.com.au/GRP-IP68-Grey-Terminal-Enclosure-250H-x-255W-x-125D (this one out of stock)

[AllBro 250x255x125.pdf](https://github.com/QUT-Motorsport/QUTMS_Driverless/files/8046750/AllBro.250x255x125.pdf)

[image]

b1n-ch1kn, 2022-02-11:
Given sizing constraints and the datasheets from the 4 test TLA options, the ones that can house the roscube are:

Fibox 278x278x180 - $119 (or the 130mm tall variant, its just $12 more)
https://au.rs-online.com/web/p/general-purpose-enclosures/1943077

Polynova 270x270x170 - $165
https://www.tradezone.com.au/product/b-r-270mm-w-x-270mm-h-x-170mm-d-square-adaptable-box-with-clear-lid-22558.html

b1n-ch1kn, 2022-02-12:
Option for Enlec 300x230x130 still on the table for secondary components. Cheap and attainable too - $42.25
https://www.tro.com.au/Polycarbonate-Enclosure-Clear-Lid-IP66-300H-x-230W-x-130D

b1n-ch1kn, 2022-02-12:
regarding chat today, i also also specced out how this would fit. 

Fibox 378x278x180 - $136.07
https://au.rs-online.com/web/p/general-purpose-enclosures/1943132

Would allow for the rear vent fan if we had the snorkel design for the fan on top

closed 2022-08-12 by b1n-ch1kn