QUT-Motorsport/QUTMS_Driverless #66: Steering actuator ballscrew cover

alistair-english, 2022-01-19:


alistair-english, 2022-03-02:
@b1n-ch1kn 's idea is we could possibly make this out of folded acrylic

alistair-english, 2022-03-08:
Potential ideas:
- 2 rubber concertinaed pieces on either side of the nut holder
- folded sheet metal cover (no mounting holes currently in chassis)
    - option here to potentially just double sided tape or something (esp if we can find one set of existing holes and tape the other side)
- or folded acryllic

alistair-english, 2022-03-08:
Want to see some ideas for next driverless meeting (15th March)

GeorgyUshakov, 2022-03-13:
2 rubber concertinaed pieces (from some research it appears they are called 'bellows') probably the better option - using an acrylic cover for the same level of dust protection seems to be more difficult given the size/shape of the steering actuator. 
Found these guys: https://www.tucks.com.au/products/bellows-protective-covers, need a quote but from the quote specification sheet it seems like we could get some customised options.
[bellowRound.pdf](https://github.com/QUT-Motorsport/QUTMS_Driverless/files/8239585/bellowRound.pdf)

- One thing to note: the bellow/accordion will have a non-zero length when fully closed so the range of the steering actuator could be reduced, need to know maximum allowable closed length.

GeorgyUshakov, 2022-03-21:
Most suitable options I've found so far (40mm wide opening and 30mm wide opening respectively; need to check dimensions of steering actuator):
https://www.ebay.com/itm/321062368556?_trkparms=amclksrc%3DITM%26aid%3D1110006%26algo%3DHOMESPLICE.SIM%26ao%3D1%26asc%3D20200818143230%26meid%3D35fb9010f10149ab9f33b12bcd90ba9e%26pid%3D101224%26rk%3D1%26rkt%3D5%26sd%3D323119785082%26itm%3D321062368556%26pmt%3D1%26noa%3D1%26pg%3D2047675%26algv%3DDefaultOrganicWeb&_trksid=p2047675.c101224.m-1

https://www.ebay.com/itm/Machinery-Rubber-30mm-Hole-Diameter-Flexible-Corrugated-Moulded-Bellow-/323119785082

Not sure exactly how these would actually be attached to the actuator though, could perhaps cut off the lips on either side and stick adhesively or with double-sided tape using the flat ends of the bellow.

GeorgyUshakov, 2022-03-24:
Go with two of the 30mm (https://www.ebay.com/itm/Machinery-Rubber-30mm-Hole-Diameter-Flexible-Corrugated-Moulded-Bellow-/323119785082)

closed 2022-07-06 by b1n-ch1kn