QUT-Motorsport/QUTMS_Driverless #91: Driverless wiring harness

b1n-ch1kn, 2022-02-10:
Needs to be made nicer + integrate rules compliance stuff like ASMS and ASSI

Prototype wiring harness, made to assist #65 

[image]

b1n-ch1kn, 2022-02-15:
RapidHarness software

b1n-ch1kn, 2022-03-30:
Had a chat with Benedetti, this could be developed as we work on a test-bench sorta thing with our mechanical and electrical systems being tested off-car first

TomFraser, 2022-04-05:
[image]

Current Status of the harness, currently transferring it to rapid harness

alistair-english, 2022-04-12:
TODO: send @johnthatfield connector order