QUT-Motorsport/QUTMS_Driverless #95: Simulate sensors in FSDS

b1n-ch1kn, 2022-02-21:
- [ ] Make a middleman node to take the stereo from FSDS and give a depth map - same specs as ZED2i camera
- [ ] bring back the old lidar code - single laserscan w/o complex ground removal

Aim to simulate sensor fusion as we dont have odom data yet

TheEppicJR, 2022-02-21:
I tried to use the robot_localization ROS package (which ends up being like 3 nodes) to act as the EKF when connected to the sim, but I haven't gotten it to work yet (a conda thing I assume) but the messages types it wants are directly compatible with what comes out of the sim as it stands.
This was the setup I was trying to follow: http://docs.ros.org/en/noetic/api/robot_localization/html/integrating_gps.html
I'm pretty sure you have to install it with rosdeps but idk it doesn't have actual install docs anywhere

b1n-ch1kn, 2022-02-21:
For this i was mostly going to try and reproduce our perception system to then look into fusion without having real odom data. But yeah we'd need some EKF for that.

The guys at RoboStack have made the `robot_localisation` package available on conda if you
`mamba install robot_localisation` (+ include it in the `conda_requirements` file in its package for the future)

[image]

closed 2022-02-23 by alistair-english