QUT-Motorsport/QUTMS_Driverless #102: 2021 vs 2022 Rules differences

b1n-ch1kn, 2022-03-03:
This is fun, 2 rules docs tabs open comparing years...

## F.4.3.1 
'22
[image]

## F.4.3.6 
'22
[image]
vs
'21
[image]

## F.5.5 
'22
[image]

## F.5.7.6
'22
[image]
vs
'21
[image]

## F.5.10
'22
[image]

## To be continued

TheEppicJR, 2022-03-04:
Guess which rule change is because of my team...

b1n-ch1kn, 2022-03-05:
> Guess which rule change is because of my team...

Which one lol

TheEppicJR, 2022-03-08:
F.8.2.4 Note