QUT-Motorsport/QUTMS_Driverless #105: DV 2.4 Autonomous System

b1n-ch1kn, 2022-03-08:
State machine :)

2020 version
[image]

2022 version
[image]

b1n-ch1kn, 2022-07-14:
[image]

TomFraser, 2022-07-27:
Hey @alistair-english @b1n-ch1kn, I would have thought the EBS Checks need to occur before the RES 'start' button is pressed so the EBS is validated before any movement is requested, as the purpose of the button is to command driving. It would be a bit weird to press the go button then the car does some checks then it goes right away.

alistair-english, 2022-07-27:
Yeah fair - how would you trigger the EBS checks? Like another button press before we send the go signal?

alistair-english, 2022-07-29:
Discussed with Bradford:

EBS Checks to be run when ASMS is switched on.

VCU is responsible for running the EBS Checks and waiting on the RES "start" button press. When EBS checks and RES start are both "ready to drive" - CAN message gets sent to autonomous systems.

closed 2022-10-26 by b1n-ch1kn