QUT-Motorsport/QUTMS_Driverless #110: LiDAR mount + cone deflector

b1n-ch1kn, 2022-03-22:
[image]
[image]
[image]
[image]

cwhit-09, 2022-05-01:
[image]

cwhit-09, 2022-05-01:
First rev done!

cwhit-09, 2022-06-19:
- Drawing needs to be checked.
- Bars need to be ordered.

b1n-ch1kn, 2022-07-12:
5005 Ally sheet
Order in square stock in plain ally, 20mm
Talk to Austin for sheets
Final review

b1n-ch1kn, 2022-07-13:
Requested quote from:
https://www.shapealuminium.net.au/aluminium-extrusion-suppliers-australia/shape-aluminium-square-box-tube-hollow/19x1.6-radius-aluminium-square-tube

closed 2022-09-23 by b1n-ch1kn