QUT-Motorsport/QUTMS_Driverless #119: Suggested documentation for setting up linux environment for Mac (including M1)

danecwalker, 2022-04-13:
[Linux VM for Mac (M1 Support).md](https://github.com/QUT-Motorsport/QUTMS_Driverless/files/8480170/Linux.VM.for.Mac.M1.Support.md)

alistair-english, 2022-04-13:
Re-posting here so can see without opening file:

# Linux VM for Mac (M1 Support)

Multipass is a simple way for Mac users to create, delete and interact with Ubuntu Virtual Machines.
It can be used in conjunction with VirtualBox or simply through terminal.

## Set up Multipass

### Install Multipass

To get started download Multipass [here](https://multipass.run/download/macos) and follow the install prompts.

### Using VirtualBox with Multipass

If you have a Mac that supports installing VirtualBox, you can link it to multipass by opening a new terminal and running:

```none
sudo multipass set local.driver=virtualbox
```

### Creating the Primary VM

To create the primary VM we will need to make sure to increase the default disk size of the VM as it defaults to 5Gb. In order to install the tools required for driverless development it is recommended to set the disk size to a minimum of 10Gb, however increasing this will ensure you have enough room for further development.

To create the VM use the following command:

```none
multipass launch --disk=20G
```

This command will download Ubuntu if it is not already available and create a new virtual machine with the name **primary** having a disk size of 20Gb.

## Using Multipass

### Using the Shell

To use the shell of a running VM, run:

```none
multipass shell <name of VM>
```

Alternatively to use the shell of the **primary** VM, run:

```none
multipass shell
```

### Start and Stop

To start running an existing VM, run:

```none
multipass start <name of VM>
```

Alternatively to start the **primary** VM, run:

```none
multipass start
```

To stop running an existing VM, run:

```none
multipass stop <name of VM>
```

Alternatively to stop the **primary** VM, run:

```none
multipass stop
```

### Listing and Deleting VM

To list the created VM's, run:

```none
multipass list
```

To delete a VM, run:

```none
multipass delete <name of VM>
```

After deleting a VM it will still show when listing VM's, however, it's status will be *Deleted*. 

To recover a deleted machine, run:

```none
multipass recover <name of VM>
```

To permanently remove a VM, run:

```none
multipass delete <name of VM>

multipass purge
```

## Additional Info

For additional usage information for Multipass visit their [docs](https://multipass.run/docs/).

danecwalker, 2022-04-13:
Added how to view desktop to run visualisation such as rviz:

# Linux VM for Mac (M1 Support)

Multipass is a simple way for Mac users to create, delete and interact with Ubuntu Virtual Machines.
It can be used in conjunction with VirtualBox or simply through terminal.

## Set up Multipass

### Install Multipass

To get started download Multipass [here](https://multipass.run/download/macos) and follow the install prompts.

### Using VirtualBox with Multipass

If you have a Mac that supports installing VirtualBox, you can link it to multipass by opening a new terminal and running:

```bash
sudo multipass set local.driver=virtualbox
```

### Creating the Primary VM

To create the primary VM we will need to make sure to increase the default disk size of the VM as it defaults to 5Gb. In order to install the tools required for driverless development it is recommended to set the disk size to a minimum of 10Gb, however increasing this will ensure you have enough room for further development.

To create the VM use the following command:

```bash
multipass launch --disk=20G
```

This command will download Ubuntu if it is not already available and create a new virtual machine with the name **primary** having a disk size of 20Gb.

## Using Multipass

### Using the Shell

To use the shell of a running VM, run:

```bash
multipass shell <name of VM>
```

Alternatively to use the shell of the **primary** VM, run:

```bash
multipass shell
```

### Start and Stop

To start running an existing VM, run:

```bash
multipass start <name of VM>
```

Alternatively to start the **primary** VM, run:

```bash
multipass start
```

To stop running an existing VM, run:

```bash
multipass stop <name of VM>
```

Alternatively to stop the **primary** VM, run:

```bash
multipass stop
```

### Listing and Deleting VM

To list the created VM's, run:

```bash
multipass list
```

To delete a VM, run:

```bash
multipass delete <name of VM>
```

After deleting a VM it will still show when listing VM's, however, it's status will be *Deleted*. 

To recover a deleted machine, run:

```bash
multipass recover <name of VM>
```

To permanently remove a VM, run:

```bash
multipass delete <name of VM>

multipass purge
```

## Accessing the Desktop

To use tools such as rviz, you will need to acces the VM's desktop. Multipass does not install a desktop environment with Ubuntu. To install the desktop and allow remote connections, first login to an instance:

```bash
multipass shell
```

once inside:

```bash
sudo apt update
sudo apt install ubuntu-desktop xrdp
```

To allow login to the remote VM create a new user or change the password of the default ubuntu user:

```bash
sudo passwd ubuntu
```

It will ask you to enter and re-enter a password. 

After that, use a tool such as Microsoft Remote Desktop to connect to the VM. 

## Additional Info

For additional usage information for Multipass visit their [docs](https://multipass.run/docs/).

closed 2022-06-28 by b1n-ch1kn