QUT-Motorsport/QUTMS_Driverless #151: Cooling system

b1n-ch1kn, 2022-07-28:
'22
[image]

So to me that reads "the reservior and cooling pipes that go to radiators need to be behind the firewall". Will need a clarification from someone who knows more.

Additionally, it's been asked that we have individual reserviors for the two radiators.

b1n-ch1kn, 2022-07-31:
Second set of reservoirs were never made - cap for reservoir needs purchasing. 
Then welding together and mounting somewhere (suggestion was above the pumps)

GeorgyUshakov, 2022-08-14:
From cooling reservoirs thread: need 2x-06 and 2x-08 Al male weld-on fittings, had 2x-06 as of 20th Jan, need to buy 2x-08 from https://www.speedflowdirect.com.au/aluminium-male-weld-on

GeorgyUshakov, 2022-10-11:
Need 2x of these caps: https://www.dicksmith.com.au/da/buy/speedparts-aeroflow-billet-radiator-cap-32mm-11b-complete-with-billet-cover-af64-5032blk/

+ 2x of these necks: https://aeroflowperformance.com/af64-0032-billet-radiator-neck-weld-on

b1n-ch1kn, 2022-11-26:
All components, minus waterjet parts, have been sorted and ready for welding

closed 2022-12-03 by b1n-ch1kn